LONGER3D/Marlin1.1.9_LGT0.3.x_Alfawise_Ux0Pro #1: Compatibility with Alfawise u30 pro

marcodori, 2019-07-26:
### Description

I've read the configuration file and i saw that the driver for both u20 and u30 pro is the same (A4988) while the u30 pro have mounted the TMC2208 driver.

There is a complete compatibility with the product or it must be reviewed?

LONGER3D, 2019-07-30:
Hello, firmware is a complete compatibility with the product,it is available for both A4988 and TMC2208. If your machine is U20_Pro, you only need unremark "U20_Pro" in Configuration.h

poisonedPawn, 2019-08-27:
I agree with @marcodori, it's confusing

LONGER3D, 2019-08-29:
Hello， TMC2208 driver board solution can be fully compatible with the existing A988 driver module, (Note: the original A988 driver work segmentation must be in 8 subdivision or 16 subdivision, that is, MS1 and MS2 pins must be pulled up) MS1 and MS2 The pin must be pulled up and the TMC2208 driver has 256 subdivision effects. In the firmware, although the motor drive selects the A4988. the driving current waveform of the TMC2208 internal processing algorithm output to the motor is 256 subdivision. The biggest advantage of this is that it reduces the burden on the master MUC, but the power consumption and heat generation of the driver IC itself are increased. If my answer is not very accurate, please correct me, we can discuss, learn, and progress together. 
  Thank you!

DeviousPenguin, 2020-01-21:
So just to confirm, if the board has a TMC2208 driver, it is best to marlin setting at A4988?

mrouillard, 2020-02-03:
Yes! The TMC2208 will do the rest by some kind of interpolation ;)

DeviousPenguin, 2020-02-09:
> Yes! The TMC2208 will do the rest by some kind of interpolation ;)

Ok, thank you. :) Just wanted to be 100% sure.