LONGER3D/Marlin1.1.9_LGT0.3.x_Alfawise_Ux0Pro #7: Upgrade to Marlin 2.x ?

SckyzO, 2020-01-23:
Hey team :)

Do you have a way to officially upgrade to Marlin 2.x?
Maybe a dev branch? I see a lot of people who would like to try this version :)
I think you would have a lot of testers considering the large community.

Thx

Mobile-Dom, 2020-01-23:
I was going to try and build my own Marlin 2.0 for it but there is some stuff that i dont know how to do or what to configure, mainly regarding the screen.

would also freaking love to see the Longer Team make ABL commands for the touchscreen, so babystepping and z probe offset.

FrancYescO, 2020-02-06:
ABL is already used in the U20 Pro, i've tried to port this on the U30 with BLTouch in my fork, basically it's working (at least while using the print with octoprint) but the LCD interaction needs some tunes...